meshtastic/meshtastic #17: Integrate info from wiki pages

sachaw, 2021-04-15:
Theres still some pages with good info on @ https://github.com/meshtastic/Meshtastic-device/wiki
Once it's done, I can remove it

jfirwin, 2021-04-17:
I'll work on this next week with @apt105 and I think we can make some serious headway.

qurm, 2021-04-26:
Hi,  I wrote the guide on the Android app wiki here:
[](https://github.com/meshtastic/Meshtastic-Android/wiki/User-Guide-for-Meshtastic-Android-App)

I'd like to move it over and update anything that needs it, and any style changes.  I still have the images on my own github repo, so should move those too.  Do we have a clear location for them?  I create a PR?

sachaw, 2021-04-26:
Just look in sidebar.js, you will see the correct location for it, feel free to submit a PR.

apt105, 2021-04-26:
I essentially did a copy and paste of that at the weekend. See https://github.com/meshtastic/Meshtastic/pull/29

Please let me know if you want to change it at all, it could do with proof reading!

qurm, 2021-04-26:
Great - will review it soon!

mkinney, 2021-11-23:
Can we close this issue?

sachaw, 2021-11-24:
Yep! I'll remove the wiki and close this now

closed 2021-11-24 by sachaw