meshtastic/meshtastic #65: Link to flashing instructions from Hardware pages?

ajorg, 2021-10-01:
I just got a T-Echo after using a T-Beam for several months and it took me a long time to figure out that the flashing instructions are completely different. I spent some time trying to get `esptool` to work before diving into the forums to find an answer.

Coming back to the docs, I see that there are flashing instructions for nRF52 devices, but I would have found them faster if there had been a link to them from the hardware page for the T-Echo.

I'd be happy to send a PR for the change, just wanted to check first if there was any objection.

jfirwin, 2021-10-01:
Hello! Thanks for bringing this up. I think adding that link mentioned to the T-Echo is a great idea. To keep things consistent we may want to check to see if we've listed the appropriate flashing instructions link for each board.

KrazyKerbalnaut, 2021-11-07:
Created PR for this issue https://github.com/meshtastic/Meshtastic/pull/81

sachaw, 2021-11-08:
@KrazyKerbalnaut if you could please resolve your merger conflict or rebase, I can close thos.

KrazyKerbalnaut, 2021-11-10:
Alright, we've updated the T-Echo hardware page to include flashing instructions links. @sachaw, if you could create a new issue and assign me to updating the rest of the hardware pages, when I have time, I can start going through and bring everything current.

closed 2021-11-10 by sachaw