meshtastic/meshtastic #172: Codespaces is out of beta

linagee, 2022-01-05:
https://meshtastic.org/docs/developers/codespaces

This page says Codespaces is in beta and you should join the beta.

GitHub's page says it is now generally available, but you need to have Team or Enterprise. ($$$) https://github.blog/changelog/2021-08-11-codespaces-is-generally-available-for-team-and-enterprise/

https://github.com/pricing

Not sure how the Docs page should change for this. Encourage people to pay $4/mo? I've never used codespaces, so I don't know if it's worth that.

garthvh, 2022-01-05:
It is much easier to get started and can easily be accessed from multiple os's so it is is a good feature. You pay $4 a month and then for CPU time which has maxed out for me at 46 cents a month.  I use it because I usually work on firmware a few days and the go back to the iOS app and the next time I need it is is still there and I don't have to care about hardware toolchains.

sachaw, 2022-01-11:
I agree, whilst the product is great, it's likely not something that needs it's own docs page. The content there is mostly useless right now anyway.

jfirwin, 2022-02-20:
Should we scrap this page and offer somewhere in the developer docs that it's a paid option that is available?

garthvh, 2022-02-20:
I say scrap it, improved docs have made it irrelevant for me.

PR #250 (2022-02-21): remove codespaces and images
closes #172
closed 2022-02-21 by jfirwin