meshtastic/meshtastic #176: Docs unclear on encryption setup

gjedeer, 2022-01-11:
I'm talking about https://meshtastic.org/docs/software/settings/channel

The docs are saying that the only possible values are 0-9, all of which are wildly insecure and based on the publically available AES key distributed in the source code.

Reading meshtastic-device source code, I've noticed that `--set-ch psk` is supposed to accept 128 and 256 bit AES keys, but I'm not sure in what format and could not find it in the documentation. Is it a 16/32 character hex string, or the baseXX-encoded string found in the channel links, or something else? I can create a PR to fix the documentation but I need to know what's correct first.

gjedeer, 2022-01-11:
I see that there's some valuable information in https://github.com/meshtastic/Meshtastic/blob/master/docs/software/settings/channel.md?plain=1#L52 but it's commented out. But it's still not clear how to set my own channel key.

jfirwin, 2022-01-11:
This is in part my fault. I was going through each settings page and really digging into source code to understand how things work life got in the way and I haven't been able to put the effort into maintaining the settings section that I have wanted.

I documented some interesting behaviors [here](https://meshtastic.discourse.group/t/encryption-clarification/3425) when I was digging into it. I think I got pulled out of my search when I found the bug which I believe has been long fixed by now.

jfirwin, 2022-01-11:
> I see that there's some valuable information in https://github.com/meshtastic/Meshtastic/blob/master/docs/software/settings/channel.md?plain=1#L52 but it's commented out. But it's still not clear how to set my own channel key.

I believe these notes are mine, they should help along side the forum post I included above.

jfirwin, 2022-01-13:
Did #178 clarify this enough that this issue can be closed or is there that should be done here?

gjedeer, 2022-01-14:
As far as I'm concerned it's good now.

closed 2022-01-14 by gjedeer