meshtastic/meshtastic #191: combine/collapse the plugins doc

mkinney, 2022-01-22:
We probably don't need two different sections:

snippets from sidebars.js

So, we probably don't need two sets of "plugin" docs.

```
        Plugins: [
          "software/plugins/plugins",
          "software/plugins/range-test-plugin",
          "software/plugins/ext-notif-plugin",
          "software/plugins/canned-message-plugin",
          "software/plugins/serial-plugin",
          "software/plugins/store-forward-plugin",
          "software/plugins/environment-plugin",
        ],
```

and

```
        Plugins: [
          "software/settings/environmental-measurement-plugin",
          "software/settings/external-notification-plugin",
          "software/settings/canned-message-plugin-settings",
          "software/settings/range-test-plugin",
          "software/settings/serial-plugin",
          "software/settings/store-and-forward-plugin",
          "software/settings/rotary-encoder-settings",
```

Perhaps we combine the these respective pages?

jfirwin, 2022-01-24:
I mentioned it on discord, but I'll say it here too. I'd like to move all material above into the settings pages. I have plans for that section. Especially with the now reformatted navigation system.

garthvh, 2022-01-24:
Do it

jfirwin, 2022-01-24:
Planning on it. Thanks! I'll figure out how to assign next time.

jfirwin, 2022-01-27:
I haven't forgotten this. Just working through how best to manage the changes on my [branch](https://github.com/jfirwin/Meshtastic/tree/consolidate-pages-to-settings)

jfirwin, 2022-02-20:
I'm closing this issue because it's being worked on but it's part of a larger refactoring that is taking a while.

closed 2022-02-20 by jfirwin