meshtastic/meshtastic #203: Privacy policy review

andrekir, 2022-01-27:
Objective: review our privacy policy to comply with Play Store requirements.

public link: https://meshtastic.org/docs/legal/privacy

minor suggested changes here: https://github.com/meshtastic/Meshtastic/commit/cb5cfc46c086d4d58cbaca724d86a46cecdc8932

- moved together main statements of user data handling to the first paragraph;
- changed "location" to "location data";

Feedback appreciated.

---

Below is a compliance checklist for privacy policies. Your privacy policy must:

- [X] 1) Reference location data and your app’s use and handling of location data
- [X] 2) Clearly label that the page is a privacy policy, for example, listed as "privacy policy" in the title or have “privacy” in the URL and body of the page
- [X] 3) Be readable in a standard browser without any plug-ins or special handlers (for example, no PDFs, no uncommon MIME types)
- [X] 4) Be available on an active, visible URL
- [ ] 5) Be non-editable
- [X] 6) Refer to your app by name
- [X] 7) Be linked on your app's Store listing page and/or within your app
- [X] 8) Have a URL that links to a single governing privacy policy in your website and/or Play Store listing rather than one that points to multiple privacy policies

TODO:
---

- [ ] remove "Edit this page" from the link. Possible?
- [ ] Review "The search engine for this website is provided by Algolia, please see their privacy policy for details of what information they collect."

Extra:
---

- [ ] add "developer information and a privacy point of contact or a mechanism to submit inquiries" like privacy@meshtastic.org?

garthvh, 2022-01-27:
I will need this soon too, would be good if the Mapbox stuff was called out as Android and WebUI only.  I will have to figure out the right phrasing for the analytics used by TestFlight when bugs are reported.

jfirwin, 2022-01-27:
I mentioned on discord but removing the button is done by adding `custom_edit_url: null` to the frontmatter yaml. I'm spinning up a branch to work on privacy policy.

jfirwin, 2022-01-27:
I found an app that is open source that puts together boilerplate privacy policies. I pasted the output in [this gist](https://gist.github.com/jfirwin/dee57d3c351152d9ef583ae5bbbb44f4) I'm sure we can create a table with the third party analytics for each app.

| Android | iOS | Web |
| :-------: | :--: | :---: |
| <ul><li>Firebase Crashalytics</li><li>Mapbox</li></ul> | <ul><li>iOS Testflight</li></ul> | <ul><li>Algolia</li></ul> |

There's definitely some work to be done on the text, but I'm happy to collaborate with anyone on getting it up to snuff.

andrekir, 2022-01-27:
@garthvh maybe we should change to app specific links? like:
- https://meshtastic.org/docs/software/android/privacy
- https://meshtastic.org/docs/software/ios/privacy

jfirwin, 2022-01-27:
We could do `legal/iOS/privacy` & `legal/android/privacy` that way it stays in the legal section.

caveman99, 2022-01-27:
breaking this out to different documents is not a good idea. things start to get messy fast and you get duplicate content. better have one document covering all. that can be grouped into chapters, highlighting different parts of the ecosystem of course.

garthvh, 2022-01-27:
Ageee that we should try and have one policy.

andrekir, 2022-01-29:
great news: the updated privacy policy was accepted! thanks for all your help!

jfirwin, 2022-02-04:
With that being fixed, do we think this issue is closed until the iOS stuff needs to be worked in?

mc-hamster, 2022-02-04:
What’s close this. Single issue tickets that have a finite state are very nice to have.

closed 2022-02-04 by andrekir