meshtastic/meshtastic #225: Community links

nuxnik, 2022-02-03:
There are other sources in the meshtastic project universe that should be displayed more prominently in the docs so they can be easily found:

1. YouTube Channel - https://www.youtube.com/meshtastic
2. Discord server - https://discord.com/invite/UQJ5QuM7vq
3. Forum - https://meshtastic.discourse.group
3. Github - https://github.com/meshtastic

Suggestion: 

1. Add these links to the footer of the page so they are visible on every page.
2. Add a "Community" page to the documentation which lists the links.

Thoughts?

jfirwin, 2022-02-03:
I love the idea of adding our links to the footer. Should be really easy to implement https://docusaurus.io/docs/api/themes/configuration/#footer-links [here](https://github.com/meshtastic/Meshtastic/blob/master/docusaurus.config.js#L94-L96)

I think we could have a `Social` column. Not sure the right header for GitHub/other documentation/legal etc, but that stuff should go down there too.

As for the Community page, I'm hesitant to add more pages as I'm trying to whittle things down to be more concise. But I think we could put these links on the front page, and in an about meshtastic section.

nuxnik, 2022-02-03:
Yeah, I agree. A community page is overkill.

PR #227 (2022-02-05): Added social links to the footer
Added social links to the footer of the page. No  

resolves #225
closed 2022-02-05 by jfirwin