meshtastic/meshtastic #226: PDF Handbook Generator

nuxnik, 2022-02-03:
# Synopisis

We should offer an PDF version of the documentation which users can download and view offline. Here are some requirements:

1. The PDF should be based on the status of the source files of the current documentation 
2. A new release of the documentation should coincide with a newly generated PDF
3. Generating a document should be based on some sort of Github action trigger
4. The document artifacts should in turn be made available for download to the documentation website: https://meshtastic.org
5. The document should have a cover sheet complete with current version number and if possible a table of contents

Documentation can be successfully generated with the following console commands. See [Docusaurus Prince PDF](https://github.com/signcl/docusaurus-prince-pdf):
```bash
npx docusaurus-prince-pdf -u https://meshtastic.org/docs/software --output meshtastic-docs.pdf
```
I was thinking the easiest way of doing this would be to put all the build tools into a docker image. This way, all the project dependencies will be encapsulated in the container and can be executed in any environment. In the container, the project would be built, spun up, then an internal process in the container could build and generate the document with Docusaurus Prince PDF. 

I'm a noob with GitHub Actions, but have a lot of experience in configuring GitLab CI Pipelines. Is is possible to trigger a docker container within GitHub or does this have to happen on an external server?

jfirwin, 2022-02-04:
Looks pretty awesome. I'd love a table of contents added as we talked about on discord.

nuxnik, 2022-02-21:
@jfirwin I'm looking into updating the library to create a TOC. PR is in the works.

jfirwin, 2022-02-21:
Awesome! I'm excited to see what you come up with. Is the TOC built using `sidebar.js`? That might be a good way to break up sections dynamically.

nuxnik, 2022-02-21:
Good point. That seems like the best place to get a complete outline of the docs. As it is, the script crawls whatever starting point you set then follows all "next" links. Now that you mention it, I might end up cannibalizing the whole script and writing a new generator that reads from the sidebar.js.

jfirwin, 2022-02-21:
I saw some interesting comments on one of the few docusaurus issues related to pdf generation. They mentioned the sidebar.js and that's what got me thinking about it.

https://github.com/facebook/docusaurus/issues/969

nuxnik, 2022-02-21:
Ah yes, that is true. Ideally, all tabs should be rendered into the documentation and added to the TOC. The selector in the generator script searches for CSS components. I suppose I could add another selector to the script which will expand the other tabs and parse the data. Haha, it's getting more and more complicated. Good call :)

jfirwin, 2022-02-22:
Sorry, work was crazy and I was doing some other documentation stuff. The tabs use a global javascript setting documented [here](https://docusaurus.io/docs/markdown-features/tabs#syncing-tab-choices). I've tried to stay consistent to the same few tab groups, but there may be some extra things lying around in the docs.

nuxnik, 2022-03-04:
Update: Sorry for the long delay. I have been very busy with work. After more research and some tests, I have decided to use [wkhtmltopdf](https://wkhtmltopdf.org) instead of Prince as the PDF generator. This has to do with licensing issues of Prince PDF, since it is a commercial product.  In the past, I have have used wkhtmltopdf in other projects professionally. It is completely open source and has an LGPL-3.0 License. I think this coincides better with the Meshtastic project.

To make it short, I am re-purposing the docusaurus-prince-pdf script for use with wkhtmltopdf. In the end, this will become it's own project. I'll rename it to docusaurus-wkhtmltopdf. The script will function exactly the same way as it's forbearer. Overview of functionality:

1. Crawl a docusaurus root URL and create a map of the site based on CSS markers of the "Next" button.
2. Read the list of URLs, iterate through them and generate a PDF.
3. I am adding TOC functionality to the project. This is inbuilt with wkhtmltopdf and is based on header "h[1-9]" HTML tags.

Here's the fork: [docusaurus-wkhtmltopdf](https://github.com/nuxnik/docusaurus-wkhtmltopdf)

nuxnik, 2022-04-06:
Sorry for the long delay. I got sick with corona. In any case, I'm back and have updated the pdf project. It can successfully generate a pdf from the documentation. You can test it by doing a pull, yarn install, then execute "npm run test". The default test script crawls the content of the meshtastic.org website, you can change this to a vercel test instance by executing:

```
node index.js -u https://vercel-instance-url/docs/getting-started
```
Additional Stuff:

1. Table of contents: I have added a very rudimentary Table of Contents. It isn't pretty. It is generated when the website is scraped, extracts the header tags and throws them into a TOC. 
2. Print CSS: In the root directory of the project there is a CSS script "print.css". This is used to modify the document for a print ready configuration. It is used for the default context and is always injected into the documents. I will be adding an extra flag to include additional CSS scripts so that it can be easily modified depending on the docusaurus instance that is being crawled.
3. Title page: Adding a title page is simple, is can be added to the cli command with the prepend option for example:

```
node index.js -u https://meshtastic.org/docs/getting-started --prepend https://meshtastic.org/docs/title-page
```
In order for this to work, we will need a "title page" to scrape.

Thoughts and Request for Comments:

I have tried to keep the dependencies to a minimum. No npm package software is necessary except wkhtmltopdf. This needs to be built into Github actions. I will be cleaning up a few things and writing some unit tests. Let me know what you think :)

sachaw, 2022-04-06:
Great work, it's starting to come together.

As you mentioned, wkhtmltopdf installation will need to be automated, there is a chance that an autoinstall npm version can be made, see:
https://github.com/sachaw/node-protoc
https://github.com/sachaw/node-protoc-gen-doc
Then the installation would not be tied to the CI and could be run easily from npm.

There are a few bits and pieces that I would change, but all preferential and can be done down the track.

We are changing up some formatting with tailwindcss in the docs, so if you need any specific styling applied to markdown etc that may make it easier, it's now possible.

garthvh, 2022-08-03:
@nuxnik can you run the docs to PDF again?  The docs have been simplified a lot, I would be interested in how it looks now.

nuxnik, 2022-08-04:
@garthvh sorry for the delay. I'm AFK on holiday for a few days. As soon as I get back, I'll regenerate the handbook. I want to get this thing finished soon anyway so we can put it in the CI pipeline :)

caveman99, 2022-11-26:
Any chance to revisit this? It looks like it would be a really good addition to the docs ...

nuxnik, 2022-11-28:
Yes definitely. I'm still working on it, although very slowly. My life caught up with me the last few months. Still a WIP, going to schedule a little more time :)

nuxnik, 2023-01-04:
Okay, I finally got some time to start rounding things up. There were a few challenges in getting this tool to work correctly. This project has grown into something much larger than I originally intended. I have included two open source software components into the project:

1. Wkhtmltopdf - generates a PDF from HTML
2. Ghostscript - Compresses PDF files.

ghostscript was integrated into the project because the resulting 300 + page meshtastic documentation pdf file was approximately 50MB or more. Ghostscript does a good job in compressing this file down to 8MB. The docusaurus-wkhtmltopdf project works as follows:

1. Crawls the docusaurus instance and creates a list of urls to add to the documentation based on "next" buttons in the documentation.
2. Creates a rudimentary table of contents
3. Downloads the single pages in list and generates a PDF.
4. Compresses the resulting PDF. This is optional with the --compress flag

Since the software dependencies were getting more complex, I decided to create a docker image. Here are the project pages including documentation:

[GitHub](https://github.com/nuxnik/docusaurus-wkhtmltopdf) - Project git repo
[npm](https://www.npmjs.com/package/docusaurus-wkhtmltopdf) - Project npm repo.
[Docker Hub](https://hub.docker.com/r/nuxnik/docusaurus-to-pdf) - The complete project in a container. This includes node, wkhtmltopdf and ghostscript.

In summary, a PDF can be created with a single command:

```
docker run --rm -v /tmp/pdf:/d2p/pdf nuxnik/docusaurus-to-pdf -u https://meshtastic.org/docs/getting-started --compress --toc
```
This command will create a table of contents, generate the pdf and compress the file. For time being I have the following challenges:

1. The tabs in the documentation do not expand for Windows/Mac. This can be tweaked by modifiying the print.css file in the project root.  Any CSS gurus here? :D
2. Page breaks aren't as clean as I would like.

As far as plugging it into the meshtastic documentation, maybe the best way would be to integrate it into Github actions and have a PDF generated for each build/release. Let me know what you think.

caveman99, 2023-01-05:
>     1. The tabs in the documentation do not expand for Windows/Mac. This can be tweaked by modifiying the print.css file in the project root.  Any CSS gurus here? :D

just adding the following code to the `src/css/custom.css` will display all content tabs during printing. Problem is, the tab headers (with highlight) are not repeated, since they are their own div. That may require an active component.

```
  @media print {
    .tabs-container div div {
        display: block;
    }
  }
```

wschnell, 2023-10-20:
I'm here - https://meshtastic.org/docs/introduction - where is the PDF?