meshtastic/meshtastic #228: 3d printable enclosures list

nuxnik, 2022-02-03:
I have suggestion concerning enclosures and anything 3d-printing related and how we can add these objects to our documentation. In order to do this, we need to create an official meshtastic thingiverse account. The purpose of this account will not be to upload new objects; instead, we will create "official" collections of enclosures that can be easily linked and added to the documentation.  

Thingiverse has a REST API which we can use to dynamically update the enclosures list and add them to the documentation: 

https://www.thingiverse.com/developers/rest-api-reference

A simple ajax GET call will suffice:
```JavaScript
const thingiverse = require('thingiverse-js');
const token = 'xxx';

thingiverse('collections/123456/things', { token }).then(res => {
  console.log(res.body);
}).catch(err => {
  console.log(err);
  console.log(thingiverse.getError(err.response));
});
```

This will make the different enclosure variations easier to find. When new ones are created, a request can be posted in the #enclosures channel or a "sticky" post on the forum. That way, they can be added to the collection.

garthvh, 2022-02-03:
Thingiverse is really broken a lot of the time, I don't think we want to take on a thingiverse dependency.

jfirwin, 2022-02-04:
3d printed cases will be talked about in the /hardware section of the docs. But I think that /showcase is going to be a place where these can really shine.

jfirwin, 2022-02-20:
@garthvh are we set on not having the thingiverse dependency? If so I think we can close this issue.

@nuxnik if you want to take a stab at a PR for 3d printed cases, I'd love to review it. It's something we definitely want, even if just for a few cases with a polished aesthetic.

garthvh, 2022-02-20:
The site often performs so badly adding a dependency does not seem like a good idea, github would be better than thingiverse. These are also designs with assorted licenses on personal accounts that are not easy to transfer to the Meshtastic Org like code in github is. I think if meshtastic manages it it should be GPL3 which is not really compatible with a lot of the CC options.

jfirwin, 2022-02-20:
I agree, I think we can showcase a couple. (tropho's come to mind) and maybe link to a thingiverse search url including meshtastic. Like this:

https://www.thingiverse.com/search?q=Meshtastic&type=things&sort=relevant

garthvh, 2022-02-20:
There is some stuff on Prusa printers too as a mirror essentially if thingiverse crashes https://www.prusaprinters.org/search/all?q=Meshtastic

jfirwin, 2022-02-20:
Oh perfect, I was going to look that one up too. Are there other sites people host these files? Or basically these two?

nuxnik, 2022-02-21:
@garthvh I'm not a Thingiverse fanboy, however I just want to throw out the following info:

1. The code for the thingiverse API is available as a node package: https://www.npmjs.com/package/thingiverse-js This package is a very simple wrapper for Ajax HTTP calls. Source code: https://github.com/makerbot/thingiverse-js
2. The license information is listed for each object on Thingiverse and most if not all fall under the CC. Since we aren't hosting these files, I don't understand how this is not compatible to the Meshtastic project. Case in point: the TTGO products are not CC or open source hardware, however we provide links to them.
3. There is no software dependency on Thingiverse. Instead the software dependency refers to the node package. This dependency is also non-breaking as it is stored in package.lock. Recovering from an unreachable API is simple. The catch block in the code fixes the issue:

```javascript
thingiverse('collections/123456/things', { token }).then(res => {
  console.log(res.body);
  // show models ...
}).catch(err => {
  console.log(err);
  // show a link instead i.e. https://www.thingiverse.com/search?q=Meshtastic&type=things&sort=relevant
});
```

The reason I suggest using the API is that it offers the advantage of being maintainable without having to manually update the docs via the typical git route - fork, change push, test, PR, merge. The only thing that needs to be done is to click the "collect" button in a thingiverse meshtastic account. 

@jfirwin let me know what route you would like to take and I'll put together a PR. In the end, I'm okay with whatever is decided by you. One more thing, https://thangs.com is also a nice place for 3d printable objects. Unfortunately it has no API... yet.

jfirwin, 2022-02-21:
We could have these display with the download cards that @sachaw put together. That way they would have a loading view/error handling. *If* we did it, I think we'd put together a small selection of cases with appropriate licensing in a collection, and link to the search urls above.(maybe as a more button with an indication it's an external link)

But then I don't know how to handle the multiple sites in that case, we'd be favoring thingiverse which might be ok, but links to the other sites should be included on the page too.

Just thinking out loud.

thebentern, 2022-02-21:
I'd actually like to see a 3D printed or hardware in general repo under Meshtastic org in Github. We could house the project files and artifacts of the models and any custom PCB antennas or boards the team wants to endorse and release to the community.

charminULTRA, 2022-05-13:
I think designs should be kept exclusively in the 3d printing websites (Thingiverse and Printables). These sites are important sources for the related design licenses, printing guidance, ratings, and commentary, and uploading .stl files to a repo or other website brings unnecessary complexity and makes things harder to find for less technical users.

I think simple links to the websites would be the easiest and simplest route.

sachaw, 2022-05-13:
My main rebuttal to that is the lack of versioning on such platforms, I agree that an up-to-date version of each case should be made available on say, the Prusa site for reach. Also, I don't see how using the repo adds any complexity to the process, designers don't even have to know how to use git, as the website has drag & drop and editing built right in.

mc-hamster, 2022-05-14:
Closing this.

Nothing stops the author of the art from submitting the same art elsewhere. It's a choice they've made to contribute their effort to this project. We can't stop someone from doing what they want. That's the spirit of choice and open source.

We welcome contributors and contributions of all forms.

closed 2022-05-14 by mc-hamster