meshtastic/meshtastic #292: Feature Request: Add simple LoRa devices to network

joshbowyer, 2022-03-21:
Im not sure if this is the right place for it, but something I would really love is to be able to have some devices on the Meshtastic network use this: https://pine64.com/product/pinephone-pinephone-pro-pindio-lora-add-on-case/

Basically, it is a LoRa chip integrated into the Pinephone, which is a full Linux smartphone. Is there a way to enable this chip to talk on the Meshtastic network? Would this require a driver/firmware? As the Pine64 LoRa project picks up speed this may be a good device to enable support for.

mc-hamster, 2022-03-21:
This is a work in progress.

https://meshtastic.org/docs/software/other/pinetab

closing as duplicate.

joshbowyer, 2022-03-21:
This is a different device and different bus