meshtastic/meshtastic #317: Firmware 1.2.59. Oled screen becomes unresponsive.

TitoSmith5107, 2022-04-14:
Please - if you just have a question, post in our [forum](https://meshtastic.discourse.group) instead. Please reserve this system for requesting changes to the Meshtastic documentation.

**Describe what section of the docs is outdated or otherwise in need of change**
Be as clear and concise as possible. Provide a URL or GitHub path to the doc you are referencing.

**Suggested Changes**
What specifically in that documentation area needs to be changed / removed / added?
If possible provide replacement text when requesting changes or new text to fill in missing documentation.

**Additional Notes**
Add any other context here.

Hey guys I made a post on the discourse a moment ago 
( https://meshtastic.discourse.group/t/oled-screen-not-working-properly-fw-1-2-59/5662 ) 

The screen on my lilygo t beam does not function for more than a few mins after power up. Have to remove battery and power back on for screen to work again… for only a few mins. If I dont remove battery the board will power up and power down but screen will remain off. Sometimes screen becomes jumbled with letter/numbers and is unresponsive. Sometimes screen goes blank and is unresponsive.  Sometimes the screen just freezes and doesnt ever dim and is unresponsive. I think its a problem with the screen sleep/ wake in the firmware. I turned on the setting “Is_always_on” via android app and turned off device sleep. Now the screen stays on but still freezes or becomes jumbled and is unresponsive after a few mins, until battery removal and repower.  I saw one other post mentioning same issue on same FW.

jfirwin, 2022-04-27:
So sorry I'm just catching this @TitoSmith5107. Are you still experiencing this behavior? I see the forum post didn't yield any responses, have you tried asking in the help channel on discord?

Issues here are meant to be for documentation issues/requests, so I'm closing this issue. But I hope you've managed to find the root cause of this. I'll post your forum post in the discord to try to get some response there. If the issue persists, the correct place to file issues with the firmware would be [meshtastic/Meshtastic-device](https://github.com/meshtastic/Meshtastic-device).

closed 2022-04-27 by jfirwin

TitoSmith5107, 2022-04-27:
> So sorry I'm just catching this ***@***.***(https://github.com/TitoSmith5107). Are you still experiencing this behavior? I see the forum post didn't yield any responses, have you tried asking in the help channel on discord?

Hey!
Its no problem man! I know you guys are busy. Yes my board is still acting up. I haven’t seen anyone who has figured out why. Just a couple posts about similar issues. Some days the oled works just fine some days it doesn’t. I cant figure out if is hardware related or firmware. I suspect firmware because the screen works perfectly some days for hours on end. Regardless of the screen functioning the board is still working as intended. Sending, receiving, gps. It is nice to be able to view the info on the oled screen as some of that info in not relayed to the phone/app. I’m just waiting for the next firmware update to see if that solves my screen issues. Thanks for getting back to me. And thanks for letting me know the correct place to post. I did post on the discord but haven’t gotten a reply last I checked. Keep up the awesome work! Im excited to see where this tech goes here soon.

Best,
-Tito

Sent from ProtonMail for iOS

jfirwin, 2022-04-27:
I reposted your forum link in the help channel on discord. Hopefully someone will have seen this behavior before.

I believe 1.2.60 was recently released, though I doubt the changes there touched whatever is going on with your screen. Best of luck! I hope you get some help, wish I knew more 🙂