meshtastic/meshtastic #456: Document button click behavior

thebentern, 2022-10-27:
Not sure if we go for unified page or section in an existing area, or if we make this hardware specific but we need to document what all of the user button click behaviors do. The problem is that specific devices like the T-Beam and T-Echo have different sets of additional buttons and button behaviors.

jfirwin, 2022-11-09:
Thanks @sigmahour I'm marking this as closed by #499

garthvh, 2022-11-09:
Those details are not correct, the BLE and AP functions are gone and there is a new triple click to send a position.

jfirwin, 2022-11-09:
@garthvh nice catch. Thanks for the clarifications. Still getting my 2.0 bearings I guess 🙂

sachaw, 2022-11-10:
I'd change it to hardware buttons instead of LoRa buttons, as they don't have anything to do with LoRa directly

sachaw, 2022-11-10:
Yeah, but hardware buttons will be obvious for even the most inept users, as they will have the context of the device anyway

sachaw, 2022-11-10:
But either way is fine

sachaw, 2022-11-12:
Can we close this as complete now?

closed 2022-11-12 by thebentern