meshtastic/meshtastic #508: Documentation Improvement Request

djjayballz, 2022-11-15:
Hi

It would be nice to have the Node-Red flow as code to copy in the MQTT docs.

Thank you

pete1450, 2022-11-17:
Having just used the existing docs; https://meshtastic.org/docs/software/mqtt/#mqtt-topics mentions the msh/ topic on the top half then starts mentioning the mesh/ topic farther down. I assume one is incorrect(I've had success with msh) and if not an explanation on the difference would be nice.

jfirwin, 2022-11-17:
Perfect! I'm hoping to dig in to this a bit more today. The PR above has a link where you can see the progress of that section.

djjayballz, 2022-11-18:
Very cool!
I added receiving, decoding and sending text messages as node-red flow examples to the docs.
We have to wait for @arduionoGP to add the missing node-red flows.

arduionoGP, 2022-11-18:
If you are setting this up for yourself take a look at my https://meshtastic.discourse.group/t/here-is-a-newb-mqtt-tutorial-for-1-3/6583/12 which I've kept more up to date till the mqtt module bugs get worked out. You will also find a bug if you try to set the default channel settings uplink/downlink via the web gui because of: https://github.com/meshtastic/web/issues/53.  Didn't post the flows because they depend on a lot of personal settings. When I get a chance, I will try to get the settings separate from the flow nodes.

djjayballz, 2022-11-19:
> If you are setting this up for yourself take a look at my https://meshtastic.discourse.group/t/here-is-a-newb-mqtt-tutorial-for-1-3/6583/12 which I've kept more up to date till the mqtt module bugs get worked out. You will also find a bug if you try to set the default channel settings uplink/downlink via the web gui because of: [meshtastic/web#53](https://github.com/meshtastic/web/issues/53). Didn't post the flows because they depend on a lot of personal settings. When I get a chance, I will try to get the settings separate from the flow nodes.

Ah, super! You can just delete your personal informations ;)
Thanks for replying

sachaw, 2022-11-20:
@djjayballz I was inspired by your writeup and made a Node-RED node available that should do all of the hard bits for you:
https://github.com/meshtastic/node-red-contrib-meshtastic/tree/master
Still got more functionality to add, but it can successfully decode the protobufs.
I plan on doing the same for n8n
I'll publish it to npm shortly

pdxlocations, 2023-09-21:
Can this be closed?

tekstrand, 2024-02-06:
If not closed, let's add details

closed 2024-02-19 by tekstrand