meshtastic/meshtastic #696: nRF52 Driver Install Clarification

pdxlocations, 2023-06-30:
On the nRF Flashing Firmware page it is stated "No driver or software install is required on any platform."
https://meshtastic.org/docs/getting-started/flashing-firmware/nrf52/

However, there are still instructions on installing drivers for Windows and Linux on the Serial Drivers pages.
https://meshtastic.org/docs/getting-started#install-serial-drivers
https://meshtastic.org/docs/getting-started/serial-drivers/nrf52

Do any platforms require serial drivers for nRF devices to load firmware or perform any other function?

garthvh, 2023-06-30:
There are certainly a lot of linux exceptions, it is not in linux mint for example.

markbirss, 2023-06-30:
@pdxlocations 

No serial driver is required to be able to use UF2

https://github.com/adafruit/Adafruit_nRF52_Bootloader
UF2 is an easy-to-use bootloader that appears as a flash drive. You can just copy .uf2-format application images to the flash drive to load new firmware. See https://github.com/Microsoft/uf2 for more information.

markbirss, 2023-06-30:
@pdxlocations 

However a serial driver would be needed for say an ESP32-S3 in order to install UF2 support as it is done via esptool which requires serial port access

https://esp32s3.com/install_uf2.html

With NRF52 you would use the SWDIO interface with a DapLink/ST-Link/J-Link/etc using the SWDIO and SWDCLK interface pins

PR #699 (2023-07-04): Reformat nRF52 drivers documentation
This PR fixes #696 

1. Specifically warn against any installation of nRF drivers on all platforms but leave options available for edge cases such as installing UF2 support.
2. Changes CH9102 driver to CH34x driver (i believe the driver name is CH34x for the CH9102 Chip?)
3. Moves Mac install instructions after the uninstall instructions so the "After Installing" Info box comes after installing.

These technologies are not in my wheelhouse so comments/edits are welcome.

[preview link](https://sandbox-git-nrf52-driver-pdxlocations.vercel.app/docs/getting-st [... cut]
closed 2023-07-04 by charminULTRA