meshtastic/meshtastic #711: LoRa Configuration Frequency Details

charminULTRA, 2023-07-06:
Device Config page for Lora:

1) Proposing to change title name from "Lora" to "LoRa Modem"
2) Page's Region settings should include the specific frequency range for all regions, not just some.
3) If possible, include or include examples for the frequencies that correspond with the channel numbers.

tekstrand, 2024-02-06:
Maybe we can link/combine with the [freq calculator](https://meshtastic.org/docs/overview/radio-settings#channel-frequency-calculator)? Having this info split across About and Configuration is confusing.

charminULTRA, 2024-03-14:
@rcarteraz That calculator is interesting, although it's not clear in the documentation what exactly the slot settings correspond to. I assume it's LoRa Channel Number, but I'm not sure why two different names were used in that case.

charminULTRA, 2024-03-14:
> > @rcarteraz That calculator is interesting, although it's not clear in the documentation what exactly the slot settings correspond to. As in, why is it called Slot if we don't have any settings called Slot?
> 
> We do have slots. We no longer use the term "LoRa Channel Number", it's now referred to as "LoRa Frequency Slot". https://meshtastic.org/docs/configuration/radio/lora/#frequency-slot It's also been updated in the android and apple apps as well as the web client. It'll be updated in the protobufs with the next breaking change.

LoRa channel number appears twice in the "Note" box here: https://meshtastic.org/docs/configuration/radio/channels/

charminULTRA, 2024-03-14:
If Frequency Slot is so important to be in Overview, it's also probably a good idea to make it much more prominent inside of that Lora Configuration page.

closed 2024-04-26 by charminULTRA