meshtastic/meshtastic #724: Documentation Change Request: /docs/development/web outdated

pdxlocations, 2023-07-13:
[Development Overview of the Web Interface](https://meshtastic.org/docs/development/web/) 
[Building](http://localhost:3000/docs/development/web/building)
and
[Managing ESP32 partitions](https://meshtastic.org/docs/development/web/esp32-partitions)

These pages are outdated and refer to an interface that is no longer implemented.  Can these pages be removed and/or replaced with docs on how to develop the current client web interface? The GitHub [Read Me](https://github.com/meshtastic/web/blob/master/README.md) would be a great start for this section.

garthvh, 2023-07-13:
Those pages are ancient, we can delete them

sachaw, 2023-07-13:
Yeah, let's rip em out

pdxlocations, 2023-07-13:
Thanks, I'll submit a PR to kill these and replace with web client dev instructions shortly.