meshtastic/meshtastic #747: Improve device comparisons

charminULTRA, 2023-07-25:
It's clear that most beginners users end up with T Echos or T Beams while NRF52 based devices offer significant benefits. We should review how we present device options to make it more clear that the extra bit of DIY can help you a bunch.

pdxlocations, 2023-08-01:
The Getting Started page has a pretty good list of the devices with basic pros and cons.  Perhaps this would be the page to expand on? 

I'm thinking we could add notes on each board such as adding to the RAK4631 description: "Great for solar and handheld devices when power conservation is important", "drag-and-drop and OTA firmware updates," and "plug-n-play add-ons such as GPS, IO breakouts, and sensors."

And/Or each device could have a list or table of advantages and disadvantages (but maybe not label it exactly that way since not having wifi may not be a disadvantage if you don't need it).

I'd like to see notes on which boards are "complete packages" and which are bare dev boards that may need a 3D printed case.

@charminULTRA I wouldn't say a T-Echo is a bad first device (though I don't particularly like it).  It's handy to not have to source a case and battery.  The Meshtastic Starter Kit is definitely my go-to recommendation but it might look a bit imposing to those without access to a 3D printer or the ability to swap out JST connectors on battery packs.

Would it be smart to link to our community's Etsy storefronts for complete devices?  I've spent more money on trial-and-error building a solar node than I would have spent on one of Garth or Keith's boxes.  On that note I'd love to find a spot to NOT recommend the RAK solar enclosure, or in a more friendly way mention that it can "extend the runtime of your radio"

I'm just thinking of the questions that come up in the Discord frequently.  It would be nice to just send them all to the docs, which are already pretty amazing if you actually read through.

It would also be wise to be forward-thinking with any edits as new devices seem to be supported frequently.

I'd be happy to help with this Issue once there is a consensus on what are where to work this stuff in.  I don't have any experience with RP2040 so I wouldn't know what say about it.

pdxlocations, 2024-01-29:
Completed in #882

closed 2024-01-29 by pdxlocations