meshtastic/meshtastic #781: Repeater role rename / clarification

charminULTRA, 2023-08-14:
Repeater role should be renamed to something like "Invisible Router" or "Transparent Router" - It's not a real repeater and seems to cause confusion for users.

Reminder to self to propose PRs for Firmware and Docs

charminULTRA, 2023-08-14:
> > Repeater role should be renamed to something like "Invisible Router" or "Transparent Router" - It's not a real repeater and seems to cause confusion for users.
> > Reminder to self to propose PRs for Firmware and Docs
> 
> I think the term repeater is fine, because that's literally what it's doing --receives and retransmits based on your rebroadcast settings. It's just folks not understanding the practical applications for using them when it comes to Meshtastic.

But since we're using a Mesh system, the term Repeater doesn't make sense. Every radio is a already repeating messages (to a hop limit of course) but repeaters don't work like that in non-Mesh radio systems, so I think we should use terminology that doesn't overlap with well-known, established radio terminology.

From Wikipedia: "An amateur radio repeater is an electronic device that receives a weak or low-level [amateur radio](https://en.wikipedia.org/wiki/Amateur_radio) signal and retransmits it at a higher level or higher power, so that the signal can cover longer distances without degradation."

pdxlocations, 2023-08-14:
My preconception of the term repeater is that the device would repeat signals from an ideal high-elevation location and there would be no user present.  Personally, I think the term works for the Meshtastic role, assuming users read the docs before blindly changing settings.  I do however see the temptation to select that role without fully understanding it.

closed 2023-08-14 by charminULTRA