meshtastic/meshtastic #807: Documentation Change Request: Position Configuration

Styne13, 2023-09-08:
In the Documentation „Position Configuration“ (https://meshtastic.org/docs/settings/config/position)
it sais:
„Position data from GPS is provided by either the radio or your paired phone.“

I‘m using iPhone App paired with TLORAv1, wich has no GPS receiver and was not able to configure the module to use phone GPS with the current documentation.

**Suggested Changes**
Add a detailed description how to configure a node without build in GPS to use GPS of paired phone.

pdxlocations, 2023-10-02:
@Styne13 Are you now able to configure your phone to share it's GPS with your module?  If so, do you have any suggestions for documentation?  Thank you!

tekstrand, 2024-02-06:
@Styne13 following up here to see if you have any suggestions, if not we can close this out.