meshtastic/meshtastic #1029: Document RAK connector Types

pdxlocations, 2024-02-19:
This is commonly asked, perhaps we can use @KeithMon 's gorgeous image in the docs
[image]

KeithMon, 2024-10-12:
@exotime @rcarteraz here's an updated version. This is for the 19007 and 19003. I'll work on the 5005 and 19001 next.

[image]
[image]

exotime, 2024-10-12:
Thanks, updated the PR with these two.

KeithMon, 2024-10-12:
@rcarteraz here are quick sheets for the 5005-O and 19001. I don't have any hands-on experience with either of these base boards so someone should review these for accuracy.

[image]

[image]

exotime, 2024-10-14:
Added those two to the PR, but I also don't have those boards. I guess someone will tell us if they are wrong.

KeithMon, 2024-10-14:
@exotime All the quick sheets have been updated to Rev 1.2. They now include absolute max and realistic max input voltages for Solar/USB.

I don't anticipate any other changes.

[image]
[image]
[image]
[image]

exotime, 2024-10-15:
Images updated to v1.2 in #1499. Thanks for the hard work @KeithMon :+1:

exotime, 2024-10-20:
I see no reason to hold up the PR over something so trivial, @rcarteraz. Can we merge this in?

KeithMon, 2024-10-23:
Thanks for finding that mistake. I fixed it:

![RAK-Wireless--RAK19007-Voltage-and-Connectors-Quick-Sheet--rev-1 2--2200x2600-Optimized](https://github.com/user-attachments/assets/ea33e010-fddb-423b-84c5-03cc
[image]
aa0d5388)
![RAK-Wireless--RAK19003-Voltage-and-Connectors-Quick-Sheet--rev-1 2--2200x2600-Optimized](https://github.com/user-attachments/assets/8e6c5343-ca64-44eb-8b04-4
[image]
c475de2b17a)

exotime, 2024-10-24:
Updated, thanks.