meshtastic/meshtastic #1036: Detailed hardware comparison table

Noki, 2024-02-22:
It would be really helpful to have a single "Table of Hardware" comparison at https://meshtastic.org/docs/hardware/devices/ showing  information for all supported devices you can buy out of the box. The table should have additional fields that are important for choosing a model.

The current page makes it hard to compare different models and brands and misses important information like the amount of flash and PSRAM that are or might become relevant with newer firmware versions and functions e.g. Store and Forward or improved routing algorithms.

Here is what I have in mind:

Brand | Model | Version | MCU | Radio | Supported Frequencies | WiFi | BT | GPS | Flash | PSRAM | Extension
--- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | ---
LILYGO | LoRa32 T3-S3 | V1.0 | ESP32-S3 | SX1262 | 868 MHz | 2.4GHz b/g/n | 5.0 | NO | 4 MB | 2 MB | SD-Card

Additional fields e.g. battery support, display resolution might also be useful and can be added int the future.

pdxlocations, 2024-04-18:
> The current page makes it hard to compare different models and brands and misses important information like the amount of flash and PSRAM that are or might become relevant with newer firmware versions and functions e.g. Store and Forward or improved routing algorithms.

What is the biggest challenge to model comparison with the current layout?
I strongly disagree with adding details that "might become relevant" with future firmware (SD Card).
I'm not at all opposed to adding information that may be relevant.

Noki, 2024-04-18:
The biggest challenge is that you can't filter or sort. With a unified table you could at least copy it to Excel or Google sheets and filter for stuff like flash or PSRAM or GPS functionality.

pdxlocations, 2024-04-18:
Interesting.  There was a push to implement a filtered search feature for hardware selection based on:
https://docusaurus.io/showcase

It proved to be too complex for me to implement.  Perhaps that would be a more appropriate feature request?  I wouldn't assume users would copy to excel to sort the options.  Another request could be a downloadable CSV of such a table.

Noki, 2024-04-19:
I would just use datatables with filters for this: https://datatables.net/extensions/fixedheader/examples/options/columnFiltering.html