meshtastic/meshtastic #1055: Clarify that serial is disabled when device is in sleep mode

nagumii, 2024-02-28:
Currently, it is unclear on the Device page that ROUTER involves sleep mode, and that sleep mode disables serial output.

Suggested fix: add the following to the serial output section: "Devices utilizing "sleep" to save power will have unreliable serial output, as it is disabled when the device enters sleep. For ROUTER role, consider switching to ROUTER_CLIENT."

tekstrand, 2024-02-28:
I saw someone in my local mesh having issues with this same issue yesterday, a few others spoke up saying they had run into the same thing, but did not know why. This explains it.

I do wonder whether "Serial disabled in sleep mode" is truly the core issue here, or if it's a lack of knowledge that Router goes in to sleep mode at all. None of my local mesh(including myself) was aware of that. And I am the one who wrote the most recent version of the device roles!

nagumii, 2024-02-28:
It needs to be clarified, absolutely.

tekstrand, 2024-03-01:
FAQ is a great idea, I hadn't thought of that.

I think the serial module section makes sense, I also wonder if adding something to the power section makes sense. One problem I've noticed we have on the docs is that some pages have **lots** of admonitions which actually makes them more difficult to digest, in my opinion. If we do want to add something to the power page, I think it would be worth considering if we can reduce overall admonitions on that page.

For the device role page, are you picturing an admonition that is something like this? Or something different?
```
### {DEVICE ROLES WITH POWER SAVING
These device roles enable power saving features which may...
```

tekstrand, 2024-03-01:
One thing that might help is adding headings to admonitions, which is what I was suggesting above with that code block.

nagumii, 2024-03-01:
> Ohhh. I didn't realize that sorry... Does that work though?

Adding to the faq is good, but there needs to be something on the device
config page imho. People won't think to look at the faq. Maybe just a
parenthetical or footnote?

tekstrand, 2024-03-01:
@nagumii we're suggesting "both and" not "either or" here.

>Ohhh. I didn't realize that sorry... Does that work though?

I think so. If nothing else, we can change the title of the admonition like it is set here for the power calc
https://meshtastic.org/docs/configuration/radio/power/#calibration-process-attribution

nagumii, 2024-03-01:
Ah, missed that. Whoops. Carry on 👍

tekstrand, 2024-03-01:
Originally I was thinking this, but I like the other way much better

[image]

tekstrand, 2024-03-01:
As a "test run" for cleaning up admonitions, I opened a PR which cleans up the Power page. I think this will help us decide what to do here on this issue.
https://github.com/meshtastic/meshtastic/pull/1069

tekstrand, 2024-03-01:
Okay, I gave it a shot. Let me know what y'all think on the PR.

tekstrand, 2024-03-02:
Thank you for your help on this @rcarteraz and @nagumii !

closed 2024-03-02 by tekstrand