meshtastic/meshtastic #1058: Document position_precision setting

jp-bennett, 2024-02-28:
The new per-channel postion_precision setting needs to be added to docs. Support is soon to land in the mobile and Python clients. In python, it's set via 
```meshtastic --ch-set module_settings.position_precision 13 --port /dev/ttyACM1 --ch-index 0```

The value of 0 means that location data is never sent over the given channel. Value 32 means it is set with full precision. The values in-between indicate the number of bits of precision to be sent. Useful values are roughly 11-16, where 11 indicates a large-ish region of plus or minus 11 Kilometers, and 16 is accurate to within a few doors down, or plus-or-minus 350 meters.

garthvh, 2024-02-28:
[image]
[image]
[image]

garthvh, 2024-02-28:
The apps will use 11-16 which is about 1/4 mile to about 9 miles, seems to be a nice range

garthvh, 2024-02-28:
The information on the iOS implementation is here https://github.com/meshtastic/Meshtastic-Apple/issues/519