meshtastic/meshtastic #1074: Client application support doesn't need to use admonitions

tekstrand, 2024-03-02:
## Current State
Every time we mention the different clients(cli, android, ios, web) we put the content into an admonition. As far as I am aware, there is not a specific reason for this. In my opinion, this makes the pages harder to read.

[image]

## Suggested State
We just make it regular text content like everything else!

tekstrand, 2024-03-02:
> pop from the page

Okay, I like your perspective here. Out of learning that I have a question:
Do you remember why it was important to have it "pop"? Were readers missing the content and asking about it even though it was on the page or something similar?

tekstrand, 2024-03-15:
Tell me more about what you mean by "Interactive Display" - would love to figure out something here

pdxlocations, 2024-04-16:
These could all be swapped out for a table.  with Supported, Not Supported, and Partially supported check marks?  I don't like how all the Android ones have: Open the Meshtastic App in the instructions.  duh.  And then it's not mentioned in the Apple tabs (maybe iphone users never close the app? :-) ) I've been wanting to standardize these all somehow.

tekstrand, 2024-04-16:
Interesting idea @pdxlocations I like it.

If we choose that route we will have another problem to contend with. We currently house most of the Python CLI docs under the CLI tab of each page, which can be quite extensive (See [channels](https://meshtastic.org/docs/configuration/radio/channels/) for an example).

I think it would make sense long term to move CLI docs to [Software > Python](https://meshtastic.org/docs/software/python/cli/) anyways. If there is general agreement on that, we can open another issue for that and it can be a prerequisite for this.

pdxlocations, 2024-04-16:
> If we choose that route we will have another problem to contend with. We currently house most of the Python CLI docs under the CLI tab of each page, which can be quite extensive (See [channels](https://meshtastic.org/docs/configuration/radio/channels/) for an example).

Looking through the config pages, it seems there are very few tabs that have any content beyond compatibility, other than the CLI examples.  A table would display all the data without user interaction.  The tabs could just go away. 

> I think it would make sense long term to move CLI docs to [Software > Python]

I'm not exactly opposed to moving all python config examples to the Software section, but we'd have to basically create a new page for each config option there or it would get pretty messy.  Mirroring what we already have in the Config pages. It may make more sense to just leave them where they are. Then we could make the Python examples a drop-down (like the new-and-improved FAQs) so only users interested in the CLI would have to see them.