meshtastic/meshtastic #1178: Linux Native Documentation Overhaul

jaredquinn, 2024-04-07:
I intend to do some cleaning up and reworking of the current documentation around the Linux Native build; both under docker and distribution package build & install.  I'm happy to make these changes after discussion here.

**Suggested Changes**

* The current section in the "Software" category seems out of place - the Linux Native build is a **node**; and I believe this section should be merged with the existing content under "Hardware/Devices".
* Expand the current "Hardware/Devices/Linux Native Devices" section; with subsections for hardware support, install/usage on OS, install/usage on docker (w/webserver content) and common configuration using config.yaml.
* Expand the docker instructions to include pulling from dockerhub along with building your own.
* Expand the docker instructions to include support for getting hardware working inside a container including GPIO support.

tekstrand, 2024-04-08:
>should be merged with the existing content under "Hardware/Devices"

Have you considered the `Development` section? The rest of the content under Hardware/Devices is truly focused on the hardware, so I don't know that it really fits there. I think it could make sense to have an overview page there, and link to the detailed docs.

jaredquinn, 2024-04-08:
> Have you considered the `Development` section? The rest of the content under Hardware/Devices is truly focused on the hardware.

I don't see why it would fit under "Development" as the content there is focused on developing around meshtastic, not running the firmware on specific hardware; which is what Linux Native is.

While development is a good reason for running Linux Native, there are plenty of use-cases here that don't include *any* development such as running an actual Linux Native node with radio.  I don't see installing on Linux to be any different to flashing the firmware on ESP32.

pedroavex, 2024-04-09:
@jaredquinn I agree with you about "I don't see installing on Linux to be any different to flashing the firmware on ESP32". In my opinion it is much more like just another hardware to be installed Meshtastic like any other with ESP32.
I can speak by myself, as a non-programmer trying to install in my Raspberry Pi 4, i had plenty of difficulties and had to cry for help in Discord. Thanks for your initiative to improve the docs about Linux Native Build!