meshtastic/meshtastic #1188: Discrepancy Regarding Paxcounter Module

moabits, 2024-04-15:
**Describe what section of the docs is outdated or otherwise in need of change**

[Documentation](https://meshtastic.org/docs/configuration/module/paxcounter/) states: "To operate the Paxcounter Module, it is mandatory to **switch _off_** both WiFi and Bluetooth in your Network and Bluetooth settings."
[iOS app](https://github.com/meshtastic/meshtastic/assets/45351488/7b6405f4-0867-40b8-b3ec-63dbfbe87225) says: "Both WiFi and Bluetooth must be **enabled** for PAX counter to work."

**Suggested Changes**

Please update _either_ the docs or the description in the app to clear up the confusion.
Does WiFi and Bluetooth have to be Enabled or Disabled in the app's settings for the Paxcounter Module to work?

pdxlocations, 2024-04-15:
Hmm, I had to turn them off to get the module to work.  @garthvh ?

caveman99, 2024-04-15:
BLE and Wifi need to be turned OFF for paxcounter to work, cause it takes over the radio of the ESP32.

garthvh, 2024-04-16:
No idea, I will go with @caveman99

pdxlocations, 2024-04-16:
Website documentation is correct

closed 2024-04-16 by pdxlocations