meshtastic/meshtastic #1259: Documentation Change Request : Linux Native Agent - systemctl and log rotation

Talie5in, 2024-05-17:
**Describe what section of the docs is outdated or otherwise in need of change**
Linux Native Hardware

**Suggested Changes**
Add After "Persistence" or part of "Persistence".
https://meshtastic.org/docs/hardware/devices/linux-native-hardware/#persistence

----

## Run Meshtastic as a service/daemon
To Setup the device to start and stop meshtasticd as as service using systemctl you can setup the service unit using the instructions below.

Create the service unit file:

Create a new file in the /etc/systemd/system/ directory with a name like meshtasticd.service.
```sudo nano /etc/systemd/system/meshtasticd.service```
Add the following content to the file:
```
[Unit]
Description=Meshtastic Daemon
After=network.target

[Service]
ExecStart=/usr/sbin/meshtasticd
Restart=always
User=nobody
Group=nogroup
StandardOutput=file:/var/log/meshtasticd.log
StandardError=file:/var/log/meshtasticd.log
Type=simple

[Install]
WantedBy=multi-user.target
```

Reload systemd to recognize the new service:
```sudo systemctl daemon-reload```

Enable the service to start on boot:
```sudo systemctl enable meshtasticd```

### Starting and Stopping the Service

Start the service:
```sudo systemctl start meshtasticd```

Check the status of the service:
```sudo systemctl status meshtasticd```

Stop the service:
```sudo systemctl stop meshtasticd```

This will give you a detailed view of the service status and any potential errors.

By following these steps, you set up a systemd service for meshtasticd that will start automatically at boot and restart if it crashes. You can manage it using the standard systemctl commands (start, stop, restart, status, etc.).

## Setup Log Rotation
To setup log rotation within the Linux Native Device and do a daily restart of the Meshtastic Daemon you can implement the below.

Create a logrotate configuration file for the meshtasticd log. Create a file named /etc/logrotate.d/meshtasticd:
```sudo nano /etc/logrotate.d/meshtasticd```

Add the following content to the file:
```
/var/log/meshtasticd.log {
    daily
    missingok
    rotate 7
    compress
    delaycompress
    notifempty
    create 0640 nobody nogroup
    sharedscripts
    postrotate
        systemctl restart meshtasticd > /dev/null 2>&1 || true
    endscript
}
```

### Test log rotation:

Test the logrotate configuration to ensure it works as expected:
```sudo logrotate -d /etc/logrotate.d/meshtasticd```

If everything looks good, force a rotation to see it in action:
``` sudo logrotate -f /etc/logrotate.d/meshtasticd```

By following these steps, you will have meshtasticd output directed to a log file with automatic log rotation configured, ensuring that your logs are managed and disk space is conserved.

lincolnpjames, 2024-05-17:
Appears the .deb release already contains the systemd service. Lets revise the above and I'll submit a PR :)

Talie5in, 2024-05-17:
This is more for non .deb installs (eg, manual compile)

Talie5in, 2024-05-19:
Need to add in something about disabling serial login on RaspbeeryPi's :/

```
sudo systemctl disable serial-getty@ttyS0.service
sudo systemctl stop serial-getty@ttyS0.service
```

Talie5in, 2024-05-21:
Alternatively, for Raspberry Pi's only... (rPi2-4 and rPi5 compatible)

```
sudo raspi-config nonint do_serial_hw 0 # Enable Serial Port (enable_uart=1)
sudo raspi-config nonint do_serial_cons 1 # Disable Serial Console
````

Can take it a step further and configure the rest of the vars..

```
sudo raspi-config nonint set_config_var dtparam=spi on /boot/firmware/config.txt # Enable SPI
sudo raspi-config nonint set_config_var dtparam=i2c_arm on /boot/firmware/config.txt # Enalbe i2c_arm

# Ensure dtoverlay=spi0-0cs is set in /boot/firmware/config.txt without altering dtoverlay=vc4-kms-v3d or dtparam=uart0
sudo sed -i -e '/^\s*#\?\s*dtoverlay\s*=\s*vc4-kms-v3d/! s/^\s*#\?\s*(dtoverlay|dtparam\s*=\s*uart0)\s*=.*/dtoverlay=spi0-0cs/' /boot/firmware/config.txt

# Insert dtoverlay=spi0-0cs after dtparam=spi=on if not already present
if ! sudo grep -q '^\s*dtoverlay=spi0-0cs' /boot/firmware/config.txt; then
    sudo sed -i '/^\s*dtparam=spi=on/a dtoverlay=spi0-0cs' /boot/firmware/config.txt
fi
```

Talie5in, 2024-06-02:
Closing as PR has been merged.

https://github.com/meshtastic/meshtastic/pull/1267

closed 2024-06-02 by Talie5in