meshtastic/meshtastic #1264: New Idea: Use Mashtastic on Esp-Now

shmuliko, 2024-05-23:
**The most use for the "alternative networks" is in emergency events in catastrophe time.**

The LoRa can connect between big neighborhoods but if a person needs to order an Ambulance, he lost time to go to the point with the device...
The ideal solution is to install LoRa Mashtastic on every street or a few buildings, **but I have another idea**:

Every Esp32 device can connected by ESP-NOW, to a small range [Officialy is 220 meters, but users succeed connected more than a kilometer].
The Esp32 devices have WiFi and BT too.

**What if can use this Technology in every building?**

The Esp32 cost 3 dollars only.
municipality organizations or communities can install many low-cost devices and publish the addresses to contact 911

When the internet is well, and electricity exists, the device can connected by the WiFi to the organization server to report the status (battery and more)
else, when a catastrophic situation, the Wi-Fi is changed to an Access-Point, for use with an application for messaging

Links: 
  An official site for the manufacture of the devices: 
    https://www.espressif.com/en/solutions/low-power-solutions/esp-now
    https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_now.html
  
  A YouTube search about esp-now:
    https://www.youtube.com/results?search_query=esp+now

garthvh, 2024-05-23:
You can bridge a meshtastic lora network via mqtt.

shmuliko, 2024-05-26:
> You can bridge a meshtastic lora network via mqtt.

Does the MQTT work when the Internet is crashed?