meshtastic/meshtastic #1302: Potential 'Meshtastic' Registration Issue in Taiwan

Oliver0804, 2024-06-20:
Some members of the Taiwanese community have noticed that 'meshtastic' could potentially be registered maliciously in Taiwan. We are providing this information to the meshtastic team. 

If assistance is needed, I am more than happy to help.

Oliver 
https://cloud.tipo.gov.tw/S282/OS0/OS0101.jsp?l6=en_US&isReadBulletinzh_TW=true

[image]

[image]

caveman99, 2024-06-20:
Meshtastic LLC is already aware of this issue, thanks for your report.

closed 2024-06-20 by caveman99

mc-hamster, 2024-06-20:
Hi Oliver,

How would you be able to assist?

Oliver0804, 2024-06-20:
https://law.moj.gov.tw/ENG/LawClass/LawAll.aspx?pcode=J0070001

According to Article 30 of the Trademark Act of Taiwan, the following grounds can be used to prevent the malicious registration of the Meshtastic trademark in Taiwan:

Article 30, Item 10:

If the trademark is identical or similar to a registered trademark or a trademark that has an earlier application date for the same or similar goods or services, and there is a likelihood of confusion among relevant consumers.
Since Meshtastic LLC has already registered or applied for the Meshtastic trademark, any identical or similar trademark application will likely confuse consumers and should be rejected.
Article 30, Item 11:

If the trademark is identical or similar to a well-known trademark or mark, and there is a likelihood of confusion among the public, or it may impair the distinctiveness or reputation of the well-known trademark or mark.
Even if Meshtastic is not registered in Taiwan, it is an internationally well-known trademark. Therefore, any application for an identical or similar trademark should be rejected to prevent confusion among the public and protect the distinctiveness and reputation of the Meshtastic trademark.
Article 30, Item 14:

If the trademark is identical or similar to the name of a well-known entity, business, or other organization, and there is a likelihood of confusion among the public.
As Meshtastic is a well-known organization internationally, any identical or similar trademark application should be rejected to avoid public confusion regarding the origin or affiliation of the products or services.
These provisions aim to protect existing trademark rights, prevent consumer confusion and misleading, and maintain the distinctiveness and reputation of the trademark. Therefore, any attempt to register the Meshtastic trademark in Taiwan maliciously should be challenged and rejected based on these grounds.

In addition, it is recommended to file an opposition against the trademark application and request the Taiwan Intellectual Property Office (TIPO) to halt the approval process. This can be done by providing evidence and arguments based on the aforementioned grounds to demonstrate the likelihood of confusion and the potential harm to the distinctiveness and reputation of the Meshtastic trademark.

If necessary, I can attempt to contact TIPO by phone.

Oliver0804, 2024-06-20:
However, I must be honest and say that filing an opposition is an attempt, and I am not certain it will stop the registration process. We can only try.