meshtastic/meshtastic #1351: Document supported GPS modules

fifieldt, 2024-07-30:
Meshtastic can automatically discover and use many different types of GPS. This means that it's generally possible to add a GPS chip to a board that doesn't have one, connect it, and it will work.

However, we don't currently have a list of the GPS chips we support. Such a list would provide guidance for users looking to add GPS to their devices.

**Suggested Changes**
Create a new section under Hardware that provides general information about GPS support in Meshtastic and lists the GPS chips currently known to be supported.

MylesBolton, 2024-08-06:
Would love this, also maybe have some more documentation in general. I think if we had more documentation, this would make learning how to build a DIY node quite accessible for most people. At the moment, the project seems to want to force people to buy pre-made nodes, unless you start digging though the repos.

MylesBolton, 2024-08-06:
allso from what I have found, any module that outputs NEMA data should work. https://www.nmea.org/

fifieldt, 2024-08-07:
Yes, provided it is configured to perform the outputting, first. Our code generally has three parts:
1. Detect the chip
2. Configure it
3. Deal with the output after its configured

Some chips will have reasonable factory defaults which mean they kinda work even if steps 1 and 2 are missing. However, I'd argue that we'd only list a chip as 'supported' if we're performing all 3 steps.

fifieldt, 2024-08-08:
Started gathering data at https://docs.google.com/spreadsheets/d/1-f9z5zx2VCYqE6ivYXm-XqtLeVZSGMKpC3G3wma6-vw/edit

contributions welcome!