meshtastic/meshtastic #1362: Potential logo copyright topic

ChrWesp, 2024-08-09:
Dear Meshtastic Community,

this is my first comment / activity within this project - so first of all really a big thank for this fantastic project and community.

To my issue: not sure if this is the right place to address this and not a real issue but I wanted to bring this into awareness.

Yesterday I found a company called Matillion which basically has an identical logo as Meshtastic (https://www.matillion.com/) - almost identical letters and even color. 
[image]

Knowing that UK / American like to sue for legal topics in such cases, I just wanted to briefly bring this to attention. I am in no way affiliated with Matillion so nothing at stake for me.

Keep up the good work!
Christian

tekstrand, 2024-08-09:
@meshtastic/admins FYI. Not sure if this is close enough to be an issue. Looks like they've been using the logo since ~September 2016 https://web.archive.org/web/20161221192721/https://www.matillion.com/

geeksville, 2024-08-09:
I'm no lawyer but my person sw eng guess is 'idk but probably fine.'  There
are **lots** of logos with an M as a key design element and they are in a
very different place.

closed 2024-08-09 by thebentern