meshtastic/meshtastic #1373: Title: High Frequency Mesh Communication Observed in Version 2.4.x

Oliver0804, 2024-08-15:
Description:
I've noticed a significant increase in mesh communication frequency in devices running version 2.4.x. Specifically, in version 2.4.1, devices like the T-Echo and T-Beam are frequently engaging in RF interactions, even when there are no other users or messages being exchanged. This behavior was not present in version 1.x, despite having more devices in the mesh network at that time.
[image]

Steps to Reproduce:

Set up two devices (e.g., T-Echo and T-Beam) with firmware version 2.4.1.
Ensure there are no active users or messages being sent in the mesh network.
Observe the frequency of RF interactions between the devices.
Expected Behavior:
The mesh communication should only occur when necessary, similar to the behavior in version 1.x.

Actual Behavior:
Devices on version 2.4.x are frequently communicating via RF, even when there is no apparent reason to do so.

Environment:

Devices: T-Echo, T-Beam
Firmware Version: 2.4.1
Previous Version with Expected Behavior: 1.x
Additional Information:
This issue could potentially lead to unnecessary power consumption and might affect the overall performance of the mesh network.

garthvh, 2024-08-15:
Are you really doing your comparison against 1.X?

Oliver0804, 2024-08-15:
> Are you really doing your comparison against 1.X?

I apologize for the confusion.
I've been mistakenly referring to version 2.3 as version 1.3. 

The correct version should be 2.3.x.