meshtastic/meshtastic #1388: Documentation Change Request

RamboRogers, 2024-08-21:
Please - if you just have a question, post in our [forum](https://meshtastic.discourse.group) instead. Please reserve this system for requesting changes to the Meshtastic documentation.

**Describe what section of the docs is outdated or otherwise in need of change**
Add North Carolina state of USA
https://github.com/meshtastic/meshtastic/edit/master/docs/community/local-groups.mdx

**Suggested Changes**
Please Add our group "Meshtastic Raleigh-Durham Users Group" to USA/North Carolina under the community 
https://discord.gg/X6Fs2kSj

**Additional Notes**
I'm excited to get this group going, there are lots of tech people in RDU.

RamboRogers, 2024-08-21:
Updated with non-expiring link https://discord.gg/uC86tef9ck

pdxlocations, 2024-11-12:
> Updated with non-expiring link https://discord.gg/uC86tef9ck

@RamboRogers, I'd be happy to add your group to the list of community groups however it appears this link is not valid or expired.

pdxlocations, 2024-11-12:
@RamboRogers It is also possible for community members to add their own listing by using the "Edit this page" button on the bottom of every docs page, fyi.

RamboRogers, 2024-11-12:
Okay thanks

pdxlocations, 2024-11-13:
> Okay thanks

Could you let me know if
You intend to close this issue and make the edit yourself
or
Leave this open for someone else to add once you've fixed the invite link

Thank you!

pdxlocations, 2025-01-06:
Invalid link

closed 2025-01-06 by pdxlocations