meshtastic/meshtastic #1490: [Feature Request]: Keyboard shortcuts documentation for t-deck.

bsouth00, 2024-10-07:
### Platform

ESP32

### Description

Running [2.3.9.f06c56a Alpha](https://github.com/meshtastic/firmware/releases/tag/v2.3.9.f06c56a) and I believe that some t-decks are being shipped from lilygo with different keyboard firmware.

For the examples below keys separated by a comma (',') indicate that the keys are pressed in sequence, releasing the previous key bafore pressing the next while keys separated by a plus ('+') indicate that the keys before is held and released after the second key is pressed.

The below keyboard shortcuts are working on my board:

- Shift+mic does NOT seem to work to enable the function layer, but the sequence "Sym,Shift+N" does
- I've been using TAB successfully since at least 2.3.7.30fbcab, maybe earlier.  The sequence is "Sym,Shift+Y"
- "Sym,Shift+N,m" - Mute (does not show the message "Notifications [Enabled/Disabled]" unless the external notification module is enabled in the module config.
- "Sym,Shift+N,g" - Toggles GPS
- "Sym,Shift+N,q" - "Quit" (is this the equivalent of an escape key?
- "Sym,Shift+J" - Also quit?
- "Sym,Shift+N,[space]" - "Position update sent" (but where was it sent? only to the primary channel?)
- "Sym,Shift+N,i" - lower screen brightness level
- "Sym,Shift+N,o" - raise screen brightness level
- After reading the rest of [This PR meshtastic/firmware#3668](https://github.com/meshtastic/firmware/pull/3668) it seems that all of the above "Sym,Shift+N" combinations also work with "Alt+C" as the Fn modifier.

Thanks to all the contributors for the frequent improvements!

31337-4554551n, 2026-01-17:
> Is there any way to send the nodeinfo message manually to force listeners to answer? I am trying to perform tests in Madrd, with many buildings, and I need to send nodeinfo messages with I needed it

I believe that's been removed from both the tdeck and techo for reasons unknown.

i5heu, 2026-02-08:
Running 2.7.15, I can confirm that alt+c does not work.