meshtastic/meshtastic #1533: Documentation Change Request - Necessary restrictions on channels using the default key

garthvh, 2024-10-28:
As meshtastic continues to grow many of the default channel restrictions being used by the public MQTT broker are going to have to be used on LoRa next as many of our lora meshes are now equivalent to the size of MQTT only meshes in the winter of 2024

Since this restrictions are likely to increase over time as traffic increases it would be good to have a page that lays out how the channels using the default key are different from private channels.

todd2982, 2025-03-07:
I'd be happy to work on this.  Can you give me some of the restrictions you are aware of so that I can start down the rabbit hole?  Are you looking for a single dedicated page, or a blurb for each module's page?

garthvh, 2025-03-07:
I would suggest the reverse, better document that the default channel mostly supports the core portnums in n order to keep messaging, discovery and position data working well on the default channel, all other features, especially those that create a lot of traffic like the detection sensor are / will be restricted on the default channel as meshtastic grows.