meshtastic/meshtastic #1563: Documentation Change Request

kaidoloor, 2024-11-09:
On https://meshtastic.org/docs/hardware/devices/lilygo/tbeam/ the flashing instructions are as follows:
```
The following process will manually place the device into the Espressif Firmware Download mode:

    Unplug the device.
    Press and hold the BOOT button.
    Plug device in.
    After 2-3 seconds, release the BOOT button.
```
This might not be correct, as this did not work for me. Instead, this works:
```
The following process will manually place the device into the Espressif Firmware Download mode:

    Unplug the device.
    Press and hold the BOOT **and POWER** buttons.
    Plug device in.
    After 2-3 seconds, release the BOOT **and POWER** buttons.
```
Pls update the docs accordingly.

Ref also to https://www.reddit.com/r/meshtastic/comments/1azj1nh/need_help_with_flashing_lilygo_ttgo_tbeam_v11/

br,
Kaido

pdxlocations, 2024-11-11:
I'm not able to test this, @rcarteraz do you have a supreme?

kaidoloor, 2024-11-16:
Guys, I would not have written if this was not an issue. I have 3 t-beams and all needed this. C'mon. You can just add that if boot down+plugin fails, then user should try boot+power+plugin.

Kaido

pdxlocations, 2024-11-16:
@kaidoloor I think the issue is that we can't add everything to the docs to address every specific situation an individual user encounters.  This could be a localized event to your setup that doesn't present as a solution to anyone else.  

Forgive us if this is a common problem/solution to flash TBeam Supreme but currently it sounds like you are the only one?

And thank you for reporting the issue and suggesting this change!  If it does look like a common solution it would be a terrific addition to the docs.