meshtastic/meshtastic #1590: [Bug?]: Duplicate entry on "Supported Hardware Overview" in documentation

digitaldisarray, 2024-11-19:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

N/A

### Description

Looking at the documentation [here](https://meshtastic.org/docs/hardware/devices/), it appears that there are two duplicate entries for the "Seeed Wio-WM1110". One of them is listed under Seeed Studio's category, while the other appears to be (incorrectly?) listed under the unPhone.

Here is a screenshot:
[image]

Figured I would open this issue to confirm that this is in fact a mistake before opening up a PR.

fifieldt, 2024-11-19:
Thanks for the report.

The Seeed Wio-WM1110 Dev Kit and the Wio Tracker 1110 Dev Kit for Meshtastic are actually two different boards with painfully similar names. However, should not be listed under unPhone :)

[image]

[image]

digitaldisarray, 2024-11-19:
I see, thank you for the quick response and the clarification! I will try and open up a pr shortly