meshtastic/meshtastic #1631: [Documentation Change Request]: show battery protection features per board

redfast00, 2024-12-17:
### Request Details

A lot of boards listed on the devices page https://meshtastic.org/docs/hardware/devices/ have a connector to connect a lipo battery. However, the device page rarely shows any details about this lipo connection or about whether they have protection features. A few I can think of:

- undervoltage protection: whether the battery will get disconnected when the voltage is too low
- proper load sharing: when the battery is charging at the same time that the microcontroller/other peripherals are using current, the battery charging IC shouldn't 'see' the current consumed by the microcontroller, otherwise it keeps charging the battery (which is bad). See also https://blog.zakkemble.net/a-lithium-battery-charger-with-load-sharing/
- for boards that have a solar panel input and have undervoltage protection: whether the board will turn on again when the battery is depleted and then recharged using the solar panel
- (maybe overcurrent protection)

This isn't really talked about at the moment, and I think we should incentivize vendors to care about this. Improper designs might damage batteries, or even cause a fire.

I'm willing to add this for some boards based on what I can see in the schematic. Would this be welcome in the docs?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pdxlocations, 2024-12-17:
I agree with @rcarteraz.  Any mistake in this kind of documentation, or an undocumented revision in a board could lead to damaged property.

tekstrand, 2024-12-17:
Agree with not adding this level to our docs. We can, should(and already do I think) make it easy to find by linking to manufacturer docs.

redfast00, 2024-12-17:
Hm, would a general warning that battery protection is needed, and what to look out for in a board be welcome then? I think that a lot of new users might not be thinking about this aspect at all; this might lead to dangerous/less optimal situations.

As far as I can see, manufacturers don't really state the battery protection features (especially not which features they're lacking); when I was looking for this, I had to search the schematic and google part numbers to see what protections it has and hasn't.

garthvh, 2024-12-17:
Most of these devices are intended to be used with flat pack batteries which have under voltage protection built in.