meshtastic/meshtastic #1661: [Bug]: Reading data with python cli from managed node

KomelT, 2025-01-07:
### Category

Serial

### Hardware

Not Applicable

### Firmware Version

2.5.15.79da236

### Description

I freshly flashed the Meshtastic node with 2.5.15.79da236 and settled it up. Additionally, I enabled is_managed: true.
After that, I tried reading data from a node with Meshtastic Python CLI and succeeded.

Why this can be done?

On [Meshtastic website](https://meshtastic.org/docs/configuration/radio/security/#managed-mode)
 states that "Enabling Managed Mode blocks client applications from accessing all radio configurations.". So this means that the blockage must be implemented on every client application and is not enforced in firmware?

tekstrand, 2025-01-07:
PR to update docs https://github.com/meshtastic/meshtastic/pull/1663

KomelT, 2025-01-07:
I still want to know why this isn't a security concern?

b8b8, 2025-01-07:
Because it's purpose is to stop your non tech savvy friend from accidentally changing settings with the android and or iOS app, not to be a security feature.

ianmcorvidae, 2025-01-07:
there is also a CLI issue to change this: meshtastic/python#615

I think that there is a case for both allowing and not allowing it with CLI, but this still isn't really a security feature since it's enforced by clients and not by the node. I can't speak to the possibility of changing it to be enforced in firmware, but figured I'd link in the issue filed against the CLI anyway.