meshtastic/meshtastic #1742: 📡 GPS Accuracy Issue Between Channel and MQTT on the Meshtastic Public Map

Supernovaes48, 2025-03-10:
Good evening everyone,

I have a question regarding the configuration of GPS accuracy and how it affects my location display. Is this a bug or a misconfiguration on my device?

In my T114 configuration, on the LongFast channel, I have set an accuracy of 729 m so that nearby users can locate me with good precision.

However, I would like to be less precise on the official Meshtastic public map. To achieve this, I have set an accuracy of 1.5 km in the MQTT settings and enabled map reporting.

Logically, I would expect:

729 m accuracy for users nearby who receive my signal directly.
1.5 km accuracy for the Meshtastic public map.
However, I have noticed that the public map ignores the accuracy set in the MQTT settings and instead uses the LongFast channel accuracy of 729 m instead of the configured 1.5 km. It seems that the accuracy setting in MQTT is not being applied.

Additionally, the only way to send my position to the public map via MQTT is to enable the "uplink" option on the LongFast channel. If I don't enable it, no position transmission occurs. The issue is that by enabling uplink, I not only transmit my location on the channel but also all channel messages, which I do not want.

This seems inconsistent—having an accuracy setting in MQTT that is not applied, while only the channel’s accuracy is used, despite map reporting being enabled and the accuracy being set in MQTT.

Is there a reason for this behavior? Is it a matter of priority between these settings?

Thanks in advance for any insights!

[image]
[image]
[image]

garthvh, 2025-03-10:
They are independent features, one uses mqtt only and no channel settings, the other uses you channel settings, sounds like you have both on.

Supernovaes48, 2025-03-10:
On the screenshots, uplink is indeed enabled. I have conducted several tests, both with and without this option.

Here are my observations:

-Uplink enabled → The position is transmitted via MQTT, but with the accuracy set in the channel. Additionally, all telemetry and messages from the channel are also sent.

-Uplink disabled, with or without map reporting in MQTT → No position is transmitted, even though logically, it should be sent via MQTT with the accuracy and map reporting settings defined in the MQTT configuration.

-Uplink enabled + Map Reporting enabled with two different accuracy settings (channel AND MQTT) → The MQTT configuration is ignored, just as when uplink is disabled. The position is always transmitted via MQTT with the channel’s accuracy.

My question:

According to the documentation, these settings can be used together, and nothing indicates that one should override the other. So, when both "uplink" and "map reporting" are enabled, which accuracy setting is actually applied: the channel’s or the one defined in MQTT?

One could assume that two different accuracy settings coexist:

1-The channel’s accuracy, which allows nearby users to locate us.
2-The accuracy defined in MQTT, which is supposed to be used for transmission via MQTT.

However, in practice, only the channel’s accuracy seems to be applied, which does not match the expected logic. It does not take the MQTT configuration into account, as if "map reporting" served no purpose without uplink.

Is this the intended behavior or a bug? I’m trying to understand.

Supernovaes48, 2025-03-10:
Thank you for your response.

Based on this observation, the location with its accuracy should be sent via MQTT to the Meshtastic MQTT server. However, in my case, my node does not do this, despite having the correct configuration.

Could it be that the Meshtastic MQTT server does not support this functionality?

I conclude that my configuration is correct:

-Location and accuracy set for the channel.
-Location and accuracy set for MQTT.

However, on the MQTT side, no data seems to be transmitted unless uplink is enabled. Map reporting does not seem to work, at least on the Meshtastic MQTT server.

Could it be that the maps only use MQTT data transmitted via uplink and not the data sent through map reporting? Is there a way to check if the node is actually transmitting map reporting via MQTT?

Thank you in advance for your help!

Supernovaes48, 2025-03-16:
Hello everyone,

I’m still looking for an explanation for my issue, and I’m wondering if it might be related to my Heltec T114 V2.

Indeed, I recently received a new G2 Station node. I applied the same configuration as on the T114, and this time, the position is correctly reported via MQTT with the accuracy value specified in the MQTT configuration.

Could there be a link with the T114? The only noticeable difference is the Internet connection: on the T114, I enabled the proxy, whereas on the G2 Station, I’m connected via Wi-Fi.

If any of you have a T114 and can run some tests to check whether you encounter the same issue, it would really help me understand the situation better.

Thanks in advance for your help!