meshtastic/meshtastic #1760: [Documentation Change Request]: Add opt-in for Web Client on ESP32

GUVWAF, 2025-03-20:
### Request Details

Would be good to add at https://meshtastic.org/docs/software/web-client/ that you now have to explicitly opt-in for the Web Client on ESP32 devices during flashing (either on the flasher website or using `—web` with the install scripts).

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

GUVWAF, 2025-03-20:
Yeah, haha just saw that part after a second read, sorry 😅

danditomaso, 2025-03-20:
Is there any chance we can have a static html page replace the web ui and simply say the web client wasn't installed. This will be a nice fallback in they choose to install it, but assumed it would be there.  More than one occasion I've come across users who expected it would be automatically installed.

todd2982, 2025-03-21:
I think that would involve running a web server on the device, and exposing more open ports that some users may have purposefully avoided doing.  I like the intent of what you are trying to do with the static page, but it may not always be wanted.

If you'd like to get started down the road of making this happen, then you could submit an issue as a feature request to the firmware repo.